MCUdude/MiniCore #288: Not able to burn bootloader to Atmega88PB

SalmanKhaja, 2024-03-06:
Hi,

I am trying use Minicore with Atmega88PB. Burning bootloader is giving following error for both both Arduino IDE version 1.8.19 and version 2.3.2.

Error from IDE version 2.3.2:
> avrdude error: invalid data {bootloader.extended_fuses} in immediate mode: cannot parse
> avrdude error: read from file {bootloader.extended_fuses} failed
> Failed chip erase: uploading error: exit status 1

Programmer: Pickit4 (programming is working when trying through Microchip studio)
MCU: Atmega88PB
LTO: Enabled
Clock: Tried internal 1MHz and 8MHz
BOD: 2.7V (3.3v power)
Bootloader: Yes (UART0)
Baudrate: Default

From some unrelated forums, it seems like issue is with fuse settings. Please look into this.

Thank you.

**### Error from IDE version 1.8.19:**

Arduino: 1.8.19 (Windows 10), Board: "ATmega88, Yes (UART0), EEPROM retained, Default, 88PB, BOD 2.7V, LTO Enabled, Internal 8 MHz"

C:\Users\Dell\AppData\Local\Arduino15\packages\MiniCore\tools\avrdude\7.2-arduino.1/bin/avrdude -CC:\Users\Dell\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.1/avrdude.conf -v -patmega88pb -cpickit4_isp -e -Ulock:w:0xff:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0b11010101:m -Ulfuse:w:0xe2:m 

avrdude: Version 7.2-arduino.1

         Copyright the AVRDUDE authors;

         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\Dell\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.1\avrdude.conf

         Using Port                    : usb

         Using Programmer              : pickit4_isp

avrdude: found CMSIS-DAP compliant device, using EDBG protocol

         AVR Part                      : ATmega88PB

         Chip Erase delay              : 10500 us

         PAGEL                         : PD7

         BS2                           : PC2

         RESET disposition             : possible i/o

         RETRY pulse                   : SCK

         Serial program mode           : yes

         Parallel program mode         : yes

         Timeout                       : 200

         StabDelay                     : 100

         CmdexeDelay                   : 25

         SyncLoops                     : 32

         PollIndex                     : 3

         PollValue                     : 0x53

         Memory Detail                 :

                                           Block Poll               Page                       Polled

           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack

           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------

           eeprom                 65    20     4    0 no        512    4      0  3600  3600 0x00 0x00

           flash                  65    10    64    0 yes      8192   64    128  4500  4500 0x00 0x00

           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00

           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00

           efuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00

           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00

           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00

           calibration             0     0     0    0 no          1    1      0     0     0 0x00 0x00

         Programmer Type : JTAG3_ISP

         Description     : MPLAB(R) PICkit 4 in ISP mode

         ICE HW version  : 3

         ICE FW version  : 1.14 (rel. 268)

         Serial number   : BUR191316283

         SCK period      : 8.00 us

         Vtarget         : 3.31 V

avrdude: AVR device initialized and ready to accept instructions

avrdude: device signature = 0x1e9316 (probably m88pb)

avrdude: erasing chip

avrdude: processing -U lock:w:0xff:m

avrdude: reading input file 0xff for lock

         with 1 byte in 1 section within [0, 0]

avrdude: writing 1 byte lock ...

avrdude: 1 byte of lock written

avrdude: verifying lock memory against 0xff

avrdude: 1 byte of lock verified

avrdude: processing -U efuse:w:{bootloader.extended_fuses}:m

avrdude fileio_imm() error: invalid data {bootloader.extended_fuses} in immediate mode: cannot parse

avrdude do_op() error: read from file {bootloader.extended_fuses} failed

avrdude done.  Thank you.

Error while burning bootloader.

SalmanKhaja, 2024-03-06:
Looks like extended_fuses settings for Atmega88 is missing in board.txt file

MCUdude, 2024-03-06:
Correct, I fixed it recently, but have not released a new version of MiniCore yet.
https://github.com/MCUdude/MiniCore/commit/d46c83fca2612e66d3059f8bb8277ee56bf20d72

Meanwhile, you can have a look at the commit and modify your boards.txt

SalmanKhaja, 2024-03-06:
Thanks dude, added 0xff as extended fuse bits. Bootloader programming is successful. But not able upload application firmware, keeps getting

> avrdude urclock_getsync() warning: attempt 1 of 10: not in sync

 I am using internal oscillator. Did I need to use any specific clock or baudrate for Atmega88PB?

SalmanKhaja, 2024-03-06:
> Thanks dude, added 0xff as extended fuse bits. Bootloader programming is successful. But not able upload application firmware, keeps getting
>
>
> I am using internal oscillator. Did I need to use any [...]

As usual issue is with hardware. Able upload user program. 

Thanks for quick help.

SalmanKhaja, 2024-03-06:
Closing this issue

closed 2024-03-06 by SalmanKhaja

kosiu, 2024-03-30:
I'm also impacted time for 3.0.2?