MCUdude/MiniCore #289: MCUSR always reports 0 at startup

smdg49, 2024-03-11:
MCUSR register always reports zero at startup. There is no way to debug what caused the last reset.

MCUdude, 2024-03-12:
Are you using a bootloader, and if yes, Urboot?

smdg49, 2024-03-12:
Bootloader ->yes (uart0)

MCUdude, 2024-03-12:
OK, and you're using the latest MiniCore version?

MCUdude, 2024-03-12:
It's most likely caused by the Urboot bootloader. However, the bootloader stores the reset cause in the R2 register. @stefanrueger do you have a proof of concept code snippet to read R2?

smdg49, 2024-03-12:
Yes

smdg49, 2024-03-12:
Do you mean general purpose register 2?

MCUdude, 2024-03-12:
Yes, see the Urboot documentation:
https://github.com/stefanrueger/urboot

I have not tried it myself.

stefanrueger, 2024-03-13:
Urboot puts the MCUSR into the processor register R2 (not the general purpose register 2), which means it must be read before the compiler does any initialisation. Below code copies it into the user-chosen variable `cpmcusr`.

```
// This variable in .noinit will not be initialised by avr-gcc's init code
uint8_t cpmcusr __attribute__ ((section(".noinit")));

// Copy r2 into global variable; needs to be done quite early on, hence choosing init0 here
void savemcusr(void) __attribute__ ((naked)) __attribute__ ((section (".init0")))  __attribute__ ((used));

void savemcusr(void) {
  __asm__ __volatile__ ("sts %0, r2\n" : "=m" (cpmcusr) :);
}
```

Urboot first keeps a copy of MCUSR in R2 and then clears MCUSR. The urboot bootloader gets invoked on external reset and normally hands over control to the application using the watchdog reset. So, you will see watchdog rather than external reset when pressing the reset button on your board. Also note that, unless you have a very clean way of powering on the device or switched off the brown-out detection, normally you will see brown-out reset rather than power-on reset.

smdg49, 2024-03-13:
Thanks for the response.
I used to be able to do
byte mcusr=MCUSR;
MCUSR=0;
Wdtdisable(): to prevent constant resetting.

I was also able to detect wether it was a external reset or wdt.

I was definitely using minicore and I must have had  some bootloader since
I was able to program using tx rx gnd reset rather than ICSP.

What changed? Did minicore have a different bootloader in the past?
What changed

stefanrueger, 2024-03-13:
Before Minicore switched to urboot it was using the optiboot bootloader, which has an ambition to [preserve the reset causes in the MCUSR as much as possible](https://github.com/Optiboot/optiboot/issues/140). This requires quite a few bytes of code in the bootloader. The urboot project doesn't see the ability to distinguish external reset from WDT reset (other than caused by the bootloader itself) as a core requirement: it prefers to spend the limited code space on other features such as autobaud detection, EEPROM r/w etc.

Best to switch to Optiboot for your requirements, @smdg49, or not use a bootloader at all (in which case you are not at mercy what it does with MCUSR). See also [this discussion from 2015](https://github.com/Optiboot/optiboot/issues/97) when optiboot showed the same MCUSR behaviour as urboot does now.

MCUdude, 2024-03-13:
> Would I lose the ability to upload through rx tx if without bootloader?

Yes, you will.

smdg49, 2024-03-13:
K. Thank you

closed 2024-03-14 by MCUdude