MCUdude/MiniCore #291: Optiboot backward compatibility

usr97629238, 2024-03-14:
Please add an entry into **programmers.txt** for the Optiboot bootloader:
```
arduino_serial.name=Arduino Serial (Optiboot)
arduino_serial.communication=serial
arduino_serial.protocol=arduino
arduino_serial.program.tool=avrdude
arduino_serial.program.tool.default=avrdude
arduino_serial.program.extra_params= -P {serial.port}
```
Sure the urclock is faster/smaller, but I for one can't be bothered to reflash the bootloader while I'm not short of those additional flash storage bytes.

MCUdude, 2024-03-14:
You can still upload using the Optiboot bootloader using the latest MiniCore version.

Avrdude's `-c urclock` is compatible with the stk500v1 protocol which Optiboot uses.

usr97629238, 2024-03-14:
Not sure what is wrong here then (the chip being a chinese knockoff?), but with Arduino IDE 1.8.19 and Optiboot 8.0 bootloader I'm getting:
```
avrdude: Version 7.2-arduino.1
...
         Using Port                    : COM3
         Using Programmer              : urclock
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
...
         Programmer Type : Urclock
         Description     : Urclock programmer for urboot bootloaders using urprotocol
         Bootloader using STK500v1 communication protocol
avrdude: AVR device initialized and ready to accept instructions
avrdude ur_initstruct() error: bootloader might be optiboot 8.0? Please use -xbootsize=<num>

avrdude avr_signature() error: unable to read signature data for part ATmega328P, rc=-1
avrdude main() error: unable to read signature data, rc=-1

avrdude done.  Thank you.

the selected serial port does not exist or your board is not connected
```

With the above modifications it flashes fine using the avrdude's arduino protocol.

MCUdude, 2024-03-14:
Where did you find the bootloader binary? Urclock is backward compatible with older Optiboot binaries that were previously bundled with MiniCore. If the chip was preloaded with a bootloader from the seller, it won't necessary work.

The reason why one would have to use -xbootsize is because the Optiboot bootloader offers no overwrite protection. If you have a binary that will fit in flash, you might accidentally overwrite the bootloader and brick the chip. When using Urclock, Avrdude reads the bootloader area and compares it with a long list of [known bootloader binaries](https://github.com/avrdudes/avrdude/blob/main/src/urclock_hash.h).

usr97629238, 2024-03-14:
It is Optiboot 8.0, flashed it myself, not from the MiniCore, but from the Optiboot project itself. Here it's working fine with the arduino protocol:
```
avrdude: Version 7.2-arduino.1
...
         Using Port                    : COM3
         Using Programmer              : arduino
         AVR Part                      : ATmega328P
...
         Programmer Type : Arduino
         Description     : Arduino for bootloader using STK500 v1 protocol
         Hardware Version: 3
         Firmware Version: 8.0
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e950f (probably m328p)
...
avrdude: 19552 bytes of flash verified

avrdude done.  Thank you.

```

MCUdude, 2024-03-14:
> It is Optiboot 8.0, flashed it myself, not from the MiniCore, but from the Optiboot project itself

Yes, that explains it.  
I have no plans to continue supporting Optiboot, other than the ones that were bundled with earlier versions of MiniCore. Urboot is better in every way. If you disagree, you're free to downgrade to MiniCore v2.x and continue to use Optiboot. the v3.0.0 was a _major_ release, which usually means breaking changes.

usr97629238, 2024-03-14:
I'm actually quite fine here, I've added those entries into programmers.txt and it's working thus far. What I thought was that it may be helpful for other users who may not know how to edit the programmers.txt file.

closed 2024-03-14 by usr97629238

WestfW, 2024-10-25:
> I have no plans to continue supporting Optiboot, other than the ones that were bundled with earlier versions of MiniCore.

That makes it rather annoying to switch between the Arduino core and MiniCore on random boards with varying bootloaders :-(

MCUdude, 2024-10-25:
`-c urclock` does support the most "common" optiboot binaries, see these links:
https://github.com/avrdudes/avrdude/blob/0b92721af9aaa686095542cc54715a3fa06b5895/src/urclock.c#L1094C2-L1140C5
https://github.com/avrdudes/avrdude/blob/main/src/urclock_hash.h

Are there other stk500v1 compatible bootloaders that are not on the list?