MCUdude/MiniCore #294: MiniCore Burn Bootloader - Not Working on M3 MacBook Pro

takide, 2024-03-30:
Burning the MiniCore bootloader does not work with the M3 MacBook Pro. I believe this issue may be caused by the USB-C to USB adaptor or some other hardware issue on Apple silicon. Previously, I was able to burn the MiniCore boot loader without any issues on a very old 2013 MacBook Pro (has USB-A ports). Burning the boot loaders works perfectly fine in windows too. Once the boot loader is burned using another computer, I can flash programs just fine using the M3 MacBook Pro.

Here is the Arduino console output when I try to burn the bootloader using the M3 MacBook Pro:
```
avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is /Users/aaron/Library/Arduino15/packages/MiniCore/hardware/avr/3.0.1/avrdude.conf
         User configuration file is /Users/aaron/.avrduderc
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbmodem1101
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega8
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom                  4    20   128    0 no        512    4      0  9000  9000 0xff 0xff
           flash                  33    10    64    0 yes      8192   64    128  4500  4500 0xff 0x00
           lfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           hfuse                   0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           lock                    0     0     0    0 no          1    1      0  4500  4500 0x00 0x00
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           calibration             0     0     0    0 no          4    1      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 version 1.x firmware

avrdude stk500_getparm() error: protocol expects OK byte 0x10 but got 0x14

avrdude stk500_getparm() error: protocol expects sync byte 0x14 but got 0x02

avrdude stk500_getparm() error: protocol expects sync byte 0x14 but got 0x10
         Hardware Version: 61
         Firmware Version: 48.48
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us
avrdude stk500_initialize() error: protocol expects OK byte 0x10 but got 0x01
avrdude main() error: initialization failed, rc=-1
        - double check the connections and try again
        - use -B to set lower the bit clock frequency, e.g. -B 125kHz
        - use -F to override this check
avrdude stk500_disable() error: protocol expects sync byte 0x14 but got 0x10

avrdude done.  Thank you.

Failed chip erase: uploading error: exit status 1
```

Here is the hardware information for the MacBook Pro having this issue:
`M3 MacBook Pro (16in Nov-2023 model, M3 Pro CPU, Sonoma 14.3.1)`

This is probably a very niche use case, but I think this should be published somewhere as a known issue.

TheZeroBeast, 2024-04-26:
I second this, I have the same model M3 MacBook Pro and have to pull out my windows machine each time I need to burn boot loaders. Would love to not have to swap machines.

MCUdude, 2024-04-26:
> I second this, I have the same model M3 MacBook Pro and have to pull out my windows machine each time I need to burn boot loaders. Would love to not have to swap machines.

@TheZeroBeast which programmer are you using that's not working with your M3-based Mac? And can you turn on verbose upload in the IDE settings and pate the _entire_ Avrdude output when a failed bootloader attempt is performed?

TheZeroBeast, 2024-04-26:
An arduino nano as ISP. I’ll try get a verbose log to share. Haven’t done any avrdudeIDE before.

MCUdude, 2024-04-26:
@TheZeroBeast but you are able to upload using your M3 Mac once the bootloader has been flashed by another computer, using the same Arduino Nano as ISP programmer?

TheZeroBeast, 2024-04-26:
@MCUdude, once I have the bootloader I connect direct with a usb to serial interface and the MacBook M3 does fine with that in Arduino.

m1chale, 2025-03-01:
im having a similar issue with my M4 Pro Macbook Pro.
It wont recognize a Nice!Nano Microcontroller connected via usb-c.
Using a linux machine I can flash new firmware without anny issues. would love to fix this issue...