MCUdude/MiniCore #297: Fails to compile Strings declared empty on Atmega328PB

Jesuslcp, 2024-04-06:
I am having issues with Minicore compiler to Atmega328PB (potentially affecting the rest of AVR's), the issue rises up when declaring empty strings like, `String inputString = "";`, this failure compiles with no returning problems and works on PB, but not when you call the function relative to the declared string at the Atmega328pb, the string does not work.
The previous failure does not show up when using the Arduino Ide compiler for Atmega328P, and the compiled binary works perfectly on the Atmega328PB.

MCUdude, 2024-04-07:
Hi!

Can you post the compiler error you're getting?

Jesuslcp, 2024-04-07:
Hey! Thanks for getting into so quick

The fact is I do not receive compiler errors , but same code works differently when I compile with Minicore or Native Arduino Ide.

With Arduino Ide the string works as expected (forcing avrdude to program the 328PB as 328P), but not when I compile with Minicore, I dont receive compile errors, but the String does not work as expected, does not capture serial data.

This is some (reduced) code you can reproduce the failure, data is received thought serial port (native serial for 328P)(Not used serial 1 option for 328PB).

`String inputString = "";  // defined variable`
 

`while (Serial.available() > 0) {  // Under loop function`

`  char inChar = (char)Serial.read();  `
`  inputString += inChar;    `          
`  delay(20); `
`}`

Jesuslcp, 2024-04-12:
Just an update into the problems found into the Atmega328PB, definitely is related to the managements of Strings, none of the strings apparently works, I did a fresh installation of minicore and problem remains the same.

The problem with the Strings disappear when I compile my 328PB as 328P with the Arduino Ide standard compiler ( However other small problems arises at other areas due to different registers identifiers).

SergIO-bel, 2024-12-15:
Hello.
I have encountered a similar problem with MiniCore and 328PB.
My part of the code is similar to the one above. Characters received from Serial are not added to the String.
I noticed that if I select the "LTO disabled" option when compiling the sketch, the problem disappears.