MCUdude/MiniCore #298: Internal Pull up error mega328

ecvary, 2024-04-13:
Internal pull up on minicore version 2.2 cannot be used?
 I made 2 minimum systems, first I made the smd mega328 version, I checked the internal serial monitor pull up on the D6 and D7 had a problem, at first I thought it was a failed chip. then I made a mega328 dip version and the results were the same, pull up pins D6 and D7 didn't work. can you help how to fix it?
[image]
[image]

MCUdude, 2024-04-13:
Please share your test code, and I'll try it on a board.

ecvary, 2024-04-13:
> Please share your test code, and I'll try it on a board.

``
#define p2 2
#define p3 3
#define p4 4
#define p5 5
#define p6 6
#define p7 7
#define p8 8
#define p9 9
#define p10 10
#define p11 11
#define p12 12

#define pA0 A0
#define pA1 A1
#define pA2 A2
#define pA3 A3
#define pA4 A4
#define pA5 A5
#define pA6 A6
#define pA7 A7

#define led 13

void setup() {
  delay(1000);
  pinMode(p2, INPUT_PULLUP);
  pinMode(p3, INPUT_PULLUP);
  pinMode(p4, INPUT_PULLUP);
  pinMode(p5, INPUT_PULLUP);
  pinMode(p6, INPUT_PULLUP);
  pinMode(p7, INPUT_PULLUP);
  pinMode(p8, INPUT_PULLUP);
  pinMode(p9, INPUT_PULLUP);
  pinMode(p10, INPUT_PULLUP);
  pinMode(p11, INPUT_PULLUP);
  pinMode(p12, INPUT_PULLUP);

  pinMode(pA0, INPUT);
  pinMode(pA1, INPUT);
  pinMode(pA2, INPUT);
  pinMode(pA3, INPUT);
  pinMode(pA4, INPUT);
  pinMode(pA5, INPUT);
  pinMode(pA6, INPUT);
  pinMode(pA7, INPUT);

  pinMode(led, OUTPUT);
  digitalWrite(led, LOW);

  Serial.begin(9600);
}

void loop() {

  int p2P = digitalRead(p2);
  int p3P = digitalRead(p3);
  int p4P = digitalRead(p4);
  int p5P = digitalRead(p5);
  int p6P = digitalRead(p6);
  int p7P = digitalRead(p7);
  int p8P = digitalRead(p8);
  int p9P = digitalRead(p9);
  int p10P = digitalRead(p10);
  int p11P = digitalRead(p11);
  int p12P = digitalRead(p12);

  int pA0P = analogRead(pA0);
  int pA1P = analogRead(pA1);
  int pA2P = analogRead(pA2);
  int pA3P = analogRead(pA3);
  int pA4P = analogRead(pA4);
  int pA5P = analogRead(pA5);
  int pA6P = analogRead(pA6);
  int pA7P = analogRead(pA7);

Serial.print(p2P);
Serial.print(p3P);
Serial.print(p4P);
Serial.print(p5P);
Serial.print(p6P);
Serial.print(p7P);
Serial.print(p8P);
Serial.print(p9P);
Serial.print(p10P);
Serial.print(p11P);
Serial.println(p12P);
delay(100);

}

ecvary, 2024-04-14:
All work when i remove analog test from sketch.
Thanks for respons.

closed 2024-04-14 by ecvary