MCUdude/MiniCore #300: Adding `-B 4` by default to `avrdude`

DRSDavidSoft, 2024-05-07:
Hi there @MCUdude and thank you for the great cores that you develop!

I've been burning the bootloader to several raw ATmega328P chips on a board I have designed, and for whatever reason, the default **Burn bootloader** option is Arduino won't work on these chips. I have to change the bit clock period to something like `-B 4` or `-B 10` for the first time programming for this to work.

Once the bootloader is successfully programmed, then the subsequent programming using the programmer works just fine.

Here's a picture of my boards:

[image]

Here's the schematics: [Custom-ATmega328p.pdf](https://github.com/MCUdude/MiniCore/files/15239069/Custom-ATmega328p.pdf)

Nothing fancy, the same chip found on the Uno and Nano. The ISP wires are not connected to anything else, they go directly to the pin headers. The length of the jumper wires is between 20cm~30cm. I'm using the USBasp programmer.

I don't know if the issue is the chips or my board design. Once I change the bit clock period, it just works:

```
avrdude -C avrdude.conf -v -p atmega328p -c usbasp  -e -Ulock:w:0xff:m -Uefuse:w:0b11111101:m -Uhfuse:w:0xd7:m -Ulfuse:w:0b11110111:m -B 4
```

(Notice the parameter added to the end.)

What I am asking is that whether we can somehow add the `-B 4` or `-B 10` to be the default when setting the fuse bits for `programmers.txt`.

I know there already exists a `USBasp slow` option as follows, but I'm not sure if the `-B32` also applies here.

```
usbasp_slow.name=USBasp slow
usbasp_slow.communication=usb
usbasp_slow.protocol=usbasp
usbasp_slow.program.protocol=usbasp
usbasp_slow.program.tool=avrdude
usbasp_slow.program.extra_params= -B32
```

It would be nice if the `-B 4` could _only_ be applied while setting the fuse bits, since we are only transferring a single byte, rather than the firmware, which wouldn't really make a difference. I don't know if the Arduino platform would support something like this, but it would be awesome if it did!

Once again, thanks for this core, and sorry for bothering you for such a minor thing, but I'd appreciate hearing your thoughts on this issue.

MCUdude, 2024-05-23:
Hi!

Applying `-B4` only to the fuse command isn't all that trivial, and the problem is that in some cases `-B4` will still be needed after the fuses have been set, because the new clock frequency is still too slow.

I'll recommend you to instead upgrade the firmware on your USBasp programmer. The latest enthusiast firmware has automatic clock speed adjustment, which means that the USBasp will clock as quick as the target can handle if no `-B` parameter have been specified.

https://github.com/dioannidis/usbasp

DRSDavidSoft, 2024-05-23:
@MCUdude Thanks for the suggestion! I'm using a self made USBasp that I built some time ago so I'm sure I can benefit from a firmware upgrade to that.

As always, thanks for the great work you do, it's always exciting to see updates to these cores!

closed 2024-05-23 by DRSDavidSoft