MCUdude/MiniCore #303: Boot time atmega168pb

muluyilmaz, 2024-06-05:
Uploading skech 8mhz internal ,bod disabled and mcu sratup 0ms with No bootlader selected  via  microchip stüdyo Vmicro.  But  any fuse  setting  , with  or  without  bootlader  the  mcu  start up  and  enter  the application software  setup  function after 3 seconds.Can not  manage  to boot  instantly.Should not the  No Bootlader  selection  start  the  mcu  instantly  without  any delay.

MCUdude, 2024-06-05:
I don't have any Visual Micro knowledge. But which version of MiniCore are you using? If you select "No bootloader", the program should start immediately. The MiniCore fuse settings adds a 65ms startup delay, but if you're experiencing a 3 second delay there might be something with your sketch. Can you show me your current fuse settings and a bare minimum sketch to try isolating the problem?

muluyilmaz, 2024-06-06:
Yeah you are right the problem was not the minicore . And it is not from the code but I found out the Vmicro Crosscompiler adds  2 sec delay in to hex file somehow for debug on Serial port .. at the Vmicro settings I set the option to Jump to source and disable debug delay options so the code started immideatly . Sorry for the wrong alert .. 

Thanks

closed 2024-06-06 by muluyilmaz