MCUdude/MiniCore #306: ATmega328PB and port PC2 problem

13LenD, 2024-06-28:
I'm working with the minicore bootloader on an ATmega 328BP version 3.0.2
Now I try to use the digital port PC2, but it does not give a normal response as I expected?
This port is externally pulled to 0V or to 5V.
So I would expect a high or low but instead I see a 16 in my serial monitor?
are there more people with this problem?
See the code used below:
`
const int Motion = PIN_PC2;       //Input motion input
const int Twilight = A3;          //Input twilight input 

void setup() {
  // initialize PWM pin Warm light as an output.
  pinMode(Motion, INPUT);
  pinMode(Twilight, INPUT);
  Serial.begin(9600);
}

// the loop function runs over and over again forever
void loop() {

  //Read the poteniometer value
  int TwilightValue = analogRead(Twilight);
   
  Serial.print("Twilight: ");
  Serial.println(TwilightValue);
  Serial.print("Motion: ");
  Serial.println(Motion);

delay(100);

}
`

MCUdude, 2024-06-28:
That's because `PIN_PC2` is defined as 16:
https://github.com/MCUdude/MiniCore/blob/0bb50f6a6648e37cbe53012c61322345847c2a03/avr/variants/standard/pins_arduino.h#L172

Instead of `Serial.println(Motion);` you should do `Serial.println(digitalRead(Motion));` instead.

13LenD, 2024-06-28:
Sorry, to come back to this!
What are the reasons that you should use digitalread here? Normally this isn't necessary, is it?
Is this due to the type of bootloader?

MCUdude, 2024-06-28:
Because `Serial.println(Motion);` will only print the value `Motion` holds (16), while `Serial.println(digitalRead(Motion));` will read the value of pin `Motion` (16). This isn't MiniCore spesific, but applies for all Arduino boards.

13LenD, 2024-06-29:
Thanks for your explanation. Learned something again!

closed 2024-06-29 by 13LenD