MCUdude/MiniCore #307: sir,i saw a problem with hfuse for atmega328p with bootloader running on internal(8MHz) and also for external(16MHz) oscillator

Sai-Free-Bird, 2024-07-17:
sir,i used your minicore board manager,thanks for sharing it in github

but the minicore **hfuse** setting for burning bootloader (with  internal(8MHz) or external(16MHz) oscillator ) has some problem

**below command is for burning bootloader for internal 8MHz oscillator for atmega328p**
avrdude -CC:\Users\mycomputer\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.2/avrdude.conf -v -patmega328p -cstk500v1 -PCOM8 -b19200 -e -Ulock:w:0xff:m -Uefuse:w:0b11111101:m -Uhfuse:w:0xd7:m -Ulfuse:w:0xe2:m 

**below command is for burning bootloader for external 16MHz oscillator for atmega328p**
avrdude -CC:\Users\mycomputer\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.2/avrdude.conf -v -patmega328p -cstk500v1 -PCOM8 -b19200 -e -Ulock:w:0xff:m -Uefuse:w:0b11111101:m -Uhfuse:w:0xd7:m -Ulfuse:w:0b11110111:m 

sir,why **hfuse** setting is **-Uhfuse:w:0xd7:m** for burning bootloader..

i cross checked the above hfuse many times.so please correct if i am mistaken...thank you sir

MCUdude, 2024-07-17:
But what is the problem? The bootloader works just fine with these fuse settings. It's not a mistake. It is called a vector bootloader. Read more here: 
https://github.com/stefanrueger/urboot

Sai-Free-Bird, 2024-07-17:
sir, when minicore -Uhfuse is 0xd7:m , the  Bit-0 : BOOTRST  will be "1".

so the minicore bootloader will be ignored as per the fuse settings..i cant understand why BOOTRST is kept "1"

i checked hfuse settings for 16mhz optiboot botloader, and BOOTRST is 0

MCUdude, 2024-07-17:
A vector bootloader does not use the bootloader section. The reset vector gets patched by Avrdude when uploading the user program. Read the Urboot README, and you'll see why.

closed 2024-07-17 by MCUdude