MCUdude/MiniCore #308: 328PB uploading trouble

Hasi123, 2024-07-26:
Hi,

I have made a custom PCB with a 328pb and I have some touble uploading through the Arduino IDE. I can upload the bootloader and the program without a bootloader through ISP, but serial programming is not working as intended. If I use an FTDI adapter, programming straight up doesn´t work at all (serial monitor does work). If I use a CP2102 adapter, it only works if I reset the board at a specific time, which is hard to hit.

I suspect this is some sort of problem with the urboot bootloader and the timing of the reset through the DTR pin. Has anyone encountered this issue?

This is the output while uploading:
`avrdude: Version 7.2-arduino.1
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

         System wide configuration file is C:\Users\HAU-NB_HLX03\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.0.2\avrdude.conf

         Using Port                    : COM5
         Using Programmer              : urclock
         Overriding Baud Rate          : 115200
avrdude urclock_getsync() warning: attempt 1 of 10: not in sync
avrdude urclock_getsync() warning: attempt 2 of 10: not in sync
avrdude urclock_getsync() warning: attempt 3 of 10: not in sync
avrdude urclock_getsync() warning: attempt 4 of 10: not in sync
avrdude urclock_getsync() warning: attempt 5 of 10: not in sync
avrdude urclock_getsync() warning: attempt 6 of 10: not in sync
avrdude urclock_getsync() warning: attempt 7 of 10: not in sync
avrdude urclock_getsync() warning: attempt 8 of 10: not in sync
avrdude urclock_getsync() warning: attempt 9 of 10: not in sync
avrdude urclock_getsync() warning: attempt 10 of 10: not in sync
avrdude urclock_recv() warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude main() error: unable to open programmer urclock on port COM5

avrdude done.  Thank you.

Failed uploading: uploading error: exit status 1`

MCUdude, 2024-07-26:
> If I use a CP2102 adapter, it only works if I reset the board at a specific time, which is hard to hit.

Sounds like there is a problem with the auto-reset circuit. Can you post a picture of your schematic?

Hasi123, 2024-07-26:
Sure:
[image]

MCUdude, 2024-07-26:
A few things:

* Are you sure that capacitor C43 is 100nF?
* Try to add a protection diode on the reset pin line [shown here](https://camo.githubusercontent.com/667d89203204c0dc275d993af539cc3f51bfc3a70fbc083bdc9abda8fb1b3358/68747470733a2f2f692e696d6775722e636f6d2f4d5858436972452e706e67). The pulses from the DTR line may cause voltage doubling when using a capacitor in series, and this may cause the ATmega328PB to enter high-voltage parallel programming mode.
* What kind of 16 MHz oscillator are you using? The 328PB is _very_ picky, so make sure this is one with low load capacitance, ideally less than 18pF.

Hasi123, 2024-07-26:
I have tried the protection diode, but there is no change.

MCUdude, 2024-07-26:
Can you try using Avrdude directly? MiniCore bundles Avrdude v7.2, but v7.3 did some changes to the auto reset logic on the host side, to be more forgiving.

Please download [Avrdude v7.3 from here](https://github.com/avrdudes/avrdude/releases/tag/v7.3) (if you're using Windows) and try to run the following command:

`avrdude.exe -curclock -patmega328pb -P COMn -b250000`

Hasi123, 2024-07-26:
I have tried the MSVC and also the mingw build, but same issue:

```
C:\Users\HAU-NB_HLX03\Downloads\avrdude-v7.3-windows-x64>avrdude.exe -curclock -patmega328pb -P COM5 -b250000
avrdude warning: attempt 1 of 10: not in sync
avrdude warning: attempt 2 of 10: not in sync
avrdude warning: attempt 3 of 10: not in sync
avrdude warning: attempt 4 of 10: not in sync
avrdude warning: attempt 5 of 10: not in sync
avrdude warning: attempt 6 of 10: not in sync
avrdude warning: attempt 7 of 10: not in sync
avrdude warning: attempt 8 of 10: not in sync
avrdude warning: attempt 9 of 10: not in sync
avrdude warning: attempt 10 of 10: not in sync
avrdude warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude error: unable to open port COM5 for programmer urclock
avrdude error: avrdude built without libserialport support; please compile again with libserialport installed

avrdude done.  Thank you.

C:\Users\HAU-NB_HLX03\Downloads\avrdude-v7.3-windows-x64>cd /d C:\Users\HAU-NB_HLX03\Downloads\avrdude-v7.3-windows_mingw-x64

C:\Users\HAU-NB_HLX03\Downloads\avrdude-v7.3-windows_mingw-x64>avrdude.exe -curclock -patmega328pb -P COM5 -b250000
avrdude warning: attempt 1 of 10: not in sync
avrdude warning: attempt 2 of 10: not in sync
avrdude warning: attempt 3 of 10: not in sync
avrdude warning: attempt 4 of 10: not in sync
avrdude warning: attempt 5 of 10: not in sync
avrdude warning: attempt 6 of 10: not in sync
avrdude warning: attempt 7 of 10: not in sync
avrdude warning: attempt 8 of 10: not in sync
avrdude warning: attempt 9 of 10: not in sync
avrdude warning: attempt 10 of 10: not in sync
avrdude warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude error: unable to open port COM5 for programmer urclock
A possible candidate serial port is:
  -P COM5 or -P cp210x
Note that above port might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

avrdude done.  Thank you.

C:\Users\HAU-NB_HLX03\Downloads\avrdude-v7.3-windows_mingw-x64>avrdude.exe -curclock -patmega328pb -P cp210x -b250000
avrdude warning: attempt 1 of 10: not in sync
avrdude warning: attempt 2 of 10: not in sync
avrdude warning: attempt 3 of 10: not in sync
avrdude warning: attempt 4 of 10: not in sync
avrdude warning: attempt 5 of 10: not in sync
avrdude warning: attempt 6 of 10: not in sync
avrdude warning: attempt 7 of 10: not in sync
avrdude warning: attempt 8 of 10: not in sync
avrdude warning: attempt 9 of 10: not in sync
avrdude warning: attempt 10 of 10: not in sync
avrdude warning: programmer is not responding; try -xstrict and/or vary -xdelay=100
avrdude error: unable to open port COM5 for programmer urclock
A possible candidate serial port is:
  -P COM5 or -P cp210x
Note that above port might not be connected to a target board or an AVR programmer.
Also note there may be other direct serial ports not listed above.

avrdude done.  Thank you.
```

MCUdude, 2024-07-26:
Can you try to bridge the 100nF capacitor and connect the DTR line directly to the reset pin on the ATmega328pb, and try Avrdude 7.3 again? Avrdude 7.3 doesn't need the 100nF cap anymore, but should still support it.

If this doesn't work, try adding `-xstrict` and/or `-xdelay=<dly_val>` and see if that helps. Also, if you have an oscilloscope, you can see what the signal looks like on the reset line.

Another thing you can try is to use the ISP programmer and Avrdude directly to inspect the flash memory to make sure the bootloader is actually there.

I'm not sure what kind of ISP programmer you've been using, but I'll assume it's a USBasp.
The Avrdude terminal (`-t`) is very powerful, and lets you do a ton of stuff like memory inspection, advanced read/write capabilities and interactive fuse calculation. Something you should definitely check out!

Try running: `avrdude.exe -cusbasp -patmega328pb -t`

And then when you get the `avrdude>` prompt, run the following command: `read flash -512 -1`. This will read the last 512 bytes from flash. Can you

Hasi123, 2024-07-26:
Uploading using ISP works just fine. I use a USBasp and the bootloader is on the chip and the fuses are set correctly. I double checked.

When I connect the DTR line directly to the reset pin, everything works flawlessly with my CP2102 and FTDI adapter as well. I can also upload the sketch via the Arduino IDE, so Avrdude 7.2 also works.

Can you somehow verify this and if you come to the same conclusion update the documentation and your reference schematics?

Hasi123, 2024-07-26:
Oh, I see the problem now. Both chips drive the DTR pin low, so the atmega won´t run as long as the reset pin is connected to DTR.

MCUdude, 2024-07-26:
> When I connect the DTR line directly to the reset pin, everything works flawlessly with my CP2102 and FTDI adapter as well. I can also upload the sketch via the Arduino IDE, so Avrdude 7.2 also works.

This reveals that something is going on with your reset circuit. a larger DTR capacitor (220n, 470n or even 1u) may solve the issue. Just make sure you have that diode there. The higher the capacitance, the higher the voltage spike may be.

> Oh, I see the problem now. Both chips drive the DTR pin low, so the atmega won´t run as long as the reset pin is connected to DTR.

Correct. But when using Avrdude 7.3 the DTR line are pulled high before attempting to communicate with the programmer

Hasi123, 2024-07-29:
I have probed the reset pin with the default 100nF cap with and without the diode.

[image]      
[image]

The datasheet mentiones a 0.2 * Vcc RESET pin threshold voltage, which is 1V in my case, however the reset pin only goes down to 1.7-1.8V. Wouldn´t this mean, that I need a smaller capacitor, so it discharges faster and therefore lower?

MCUdude, 2024-07-29:
If the voltage isn't pulled "down enough" the RESET pin has for some reason a too strong pullup. This may be the internal pullup resistor in the chip. Where did you buy the chip?

You'll need a higher value capacitor (470nf or perhaps 1uF) to achieve an impedance that's low enough to overcome the stronger pullup resistor

Hasi123, 2024-07-29:
Actually the problem is, that the serial chip only has a 3.3V voltage level on the DTR pins and due to the inline cap, the voltage will theoretically never go below 5-3.3=1.7V, which is exactly what i see on my scope.

My final solution is to power the 328pb with 3.3V (which is a bit out of spec at 16MHz, but it works) while programming. What do you think?

MCUdude, 2024-07-29:
> Actually the problem is, that the serial chip only has a 3.3V voltage level on the DTR pins and due to the inline cap, the voltage will theoretically never go below 5-3.3=1.7V, which is exactly what I see on my scope.

Oh, that's a good point! The ATmega328P (non-PB) has a more forgiving reset threshold level, or it would work with a 328P, but not a 328PB. a 5V DTR signal or a logic buffer like the 74LVC1G125 would also work.

It's a bit tricky. And most of the problems reported here are users that are having problems with the 328PB. The older chips are way more forgiving. It's also very annoying when chips don't follow their specs, like triggering HV mode. I'll update the reference setup schematics in this repo so they all have a protection diode.

Hasi123, 2024-07-29:
Ok, thanks for you help! You only have to update your reference schematic for the 328PB, since the older chips works just fine without the diode.

closed 2024-07-29 by Hasi123