MCUdude/MiniCore #309: 328PB Challenges

lucas-andersen, 2024-07-27:
Hi MCUdude, hoping you can help me get my 328pb design up and running. Schematic is attached. I'm attempting to upload the blink arduino sketch using the AVRISP XPII programmer but getting the error message attached.

I've played around a bunch at this point and seen a few different error messages and even received a success message during the upload process although D17 is not blinking so something is clearly not working. I probed the reset line during the flashing process and it is being pulled from 4.8V to 0.6V for about 50ms. One of my concerns was the 1K pullup on RST being too small. I DNP'd the capacitors on my crystal which is TSX-3225 16.0000MF09Z-AC0 (9pF load capacitance) just based on datasheet calculations and comparing my design with the 328pb to some of AVR's example layouts on their hardware design considerations app note. That said, my understanding is that it should default to the internal 8MHz resonator upon first programming, so this shouldn't matter? Let me know if you have any thoughts. Thanks!
[image]
[Error Messages.txt](https://github.com/user-attachments/files/16400925/Error.Messages.txt)

MCUdude, 2024-07-27:
What kind of crystal oscillator are you using? The load capacitors are marked as DNP. Is this correct? And why did you design your reset circuitry like that? The protection diode isn't correctly placed, and why told you that a 1k pull up resistor is what's common?

Remove the TVS diode and the capacitor on the reset line, and replace the pull up resistor with a 10k one.

lucas-andersen, 2024-07-27:
I'm using TSX-3225 16.0000MF09Z-AC0 (9pF load capacitance). I explained why I DNP'd the load capacitors in my note above. My reset circuitry design came directly form AVR's MCU hardware design app note. 
[image]

MCUdude, 2024-10-10:
Well, your hardware setup looks correct. If you have an ocilloscope, you can try to measure the XTAL1 pin to see if the oscillator is working or not.

lucas-andersen, 2024-10-10:
I ended up getting it to work. Replaced 1K pull-up with 10K and pulled off
the switch. Thanks for the tips. Weird that I was seemingly following AVR’s
hardware recommendations. Maybe it was just the pull-up and leaving the
switch would have been fine.

MCUdude, 2024-10-10:
Great to hear you figured it out!

closed 2024-10-10 by MCUdude