MCUdude/MiniCore #313: Serial.print send gibberish text

KLevi0, 2024-08-13:
Hi! First of all, I would like to thank you for your work! It was very easy to burn the bootloader, however after testing different baud rates I still can't get any usable text.
`
23:32:20.581 -> I��i�5��۱�W5
23:32:22.490 -> ����W5�I[��W5
23:32:24.374 -> I۱��5��[��W5�I[��W5�I[���5��۱��5
23:32:29.162 -> I[���5
23:32:30.068 -> �[���5
23:32:31.018 -> �۱��5
23:32:32.006 -> �۱iW5��۱i�5�I��i�5��[��5
23:32:39.437 -> I���5
23:32:40.387 -> `

Settings:
Atmega328p
Bootloader: yes
Internal 8MHz
EEPROM retained
LTO enabled

MCUdude, 2024-08-13:
At which baud rate are you printing text to your PC?

KLevi0, 2024-08-13:
I tried from 300 to 57600 (and a couple faster).
I got it working after changing the setting in tools to 115200 and I also reburned the bootloader.
I didn't notice the setting in tools for baud rate.

closed 2024-08-13 by KLevi0