MCUdude/MiniCore #316: Question: How do I add a custom clock option for ATMega328P in the Arduino IDE menu?

sidharthbhatia, 2024-09-17:
Hi @MCUdude,

Thanks a lot for the awesome core! It really simplifies a ton of things.
I am using the core version `3.0.2` and I want to add a new clock option to the `Clock` selection menu for the ATMega328P. I want to call it `2MHz_16MHz_external`. Using this, I intend to activate the `CKDIV8` fuse bit to run the MCU at 2MHz with the 16MHz external oscillator found on most ATMega328P Arduino boards.
I followed the changes you did in your recent commit [Add 9.216 MHz clock](https://github.com/MCUdude/MiniCore/commit/d732671a3647b593172dc4a9d71cdaf9c23be3b8). Here are the changes that I did to the `boards.txt` file:

```
328.menu.clock.2MHz_16MHz_external=External 2 16 MHz
328.menu.clock.2MHz_16MHz_external.upload.default_speed=9600
328.menu.clock.2MHz_16MHz_external.bootloader.low_fuses=0b0111{bootloader.cksel_bits}
328.menu.clock.2MHz_16MHz_external.build.clkpr=
328.menu.clock.2MHz_16MHz_external.build.f_cpu=2000000L
```

Even after doing these changes and restarting the Arduino IDE, I am not able to see this option in the menu. I also tried changing the name of an existing clock option, e.g. changing `External 2 MHz` to `External 2 MHz Osc` but no change seems to be reflecting in the menu. However, when I change the fuse bits for the `External 2 MHz` and burn the bootloader to the MCU, I am getting the correct fuse values getting written to the MCU.

Can you please let me know what else I need to do here?

As an additional info, I also tried deleting the `cache` and `staging` folders, which didn't work.

MCUdude, 2024-09-17:
Hi!

I'm not sure what the problem is. I've installed MiniCore though the boards manager, and when editing the boards.txt located in the following path on my mac, the `2 16 MHz external` option shows up without any issues:

`/Users/hans/Library/Arduino15/packages/MiniCore/hardware/avr/3.0.2/boards.txt`

[image]

sidharthbhatia, 2024-09-18:
Hmm. That's strange. I tried removing and reinstalling MiniCore via the Arduino IDE boards manager, but I didn't get any success doing that as well. I could try uninstalling and reinstalling the Arduino IDE to see if that solves it.

MCUdude, 2024-10-10:
Well, there's not much more I can do at this point.

closed 2024-10-10 by MCUdude