MCUdude/MiniCore #319: Can't burn bootloader to newer 328PB

mountainmannm, 2024-10-05:
Hello, 
I got a tray of 50 328PB's in the mail recently. I'm trying to use them and it seems that I can't burn the bootloader. I get the error below. My setup still works fine with other 328PB's that I've got so I think there's something different about the new batch of chips. Seems like something in the mini core library may need to be updated? Any help would be much appreciated. Thanks!

Error:
Error while burning bootloader.
avrdude error: Yikes!  Invalid device signature.
avrdude error: expected signature for ATmega328PB is 1E 95 16
        Double check connections and try again, or use -F to override
        this check.

mountainmannm, 2024-10-07:
False alarm! It was a hardware issue that followed me to two separate PCB layouts. I didn't have the 5v rail connected to the 328pb.

MCUdude, 2024-10-07:
Great to hear you figured it out!

closed 2024-10-07 by MCUdude