MCUdude/MiniCore #322: Arduino Nano flashed with internal 8mhz - Serial Problems

antodima90, 2024-10-25:
Hi, I used the MiniCore boards to flash the bootloader version that provides 8mhz internally on an Arduino Nano.
I flashed using an Arduino UNO connected via ISP to the Arduino Nano that hosts the new bootloader. Everything happened correctly also because with some tools I read the Fuse bits setting and they are like this:

> 10:18:33.740 -> LFuse = 0xE2
10:18:33.740 -> HFuse = 0xD7
10:18:33.740 -> EFuse = 0xFD

The crucial point is that I'm having big problems with the Serial. In fact, even if I try different baud rates of 9600 or 4800 etc., unfortunately the Serial prints strange characters as if there was a discrepancy between the two speeds of the microcontroller and the serial monitor, even though they are set the same for both. Obviously when I load a sketch, I set "ATMEGA328" as the board and "Internal 8Mhz" as the clock. I leave the Baudrate "few" (in italian appears "pochi") even if I tried to change it without solving anything.
How could this problem be solved? Thanks.

MCUdude, 2024-10-25:
The problem is that the internal oscillator isn't very accurate, and would need calibration in order to work. For this you'll have to use the `OSCCAL` register. Try this piece of code to see what the ideal OSCCAL value would be:

```c
 OSCCAL = OSCCAL - 10;

while(OSCCAL < 0x7F) {
  OSCCAL++;
  Serial.printf("New OSCCAL value is 0x%02x\n", OSCCAL);
  delay(1000);
}
```

At the start of every program you'll have to write the correct value to the OSCCAL register:
`OSCCAL = [some_value];`

morpheus19900, 2024-10-25:
Thank you so much I solved it!
Initially your code did not print any value. I modified it like this and I found readable values ​​to set OSCCAL in order to have a clear reading of the Serial. I set it for 40 values.

Thanks again!

```cpp
void setup() {
  delay(1000);
  
  Serial.begin(1200);  
  delay(1000);
  
  uint8_t original_osccal = OSCCAL;
  uint8_t start_value = original_osccal - 20;  
  
  for(uint8_t i = 0; i < 40; i++) {  // Test di 40 valori
    uint8_t test_value = start_value + i;
    OSCCAL = test_value;
    delay(1000);  
    
    for(int j = 0; j < 5; j++) {
      Serial.write('=');
    }
    Serial.write(' ');
    Serial.write('0' + (test_value / 100));     
    Serial.write('0' + ((test_value / 10) % 10)); 
    Serial.write('0' + (test_value % 10));       
    Serial.write(' ');
    for(int j = 0; j < 5; j++) {
      Serial.write('=');
    }
    Serial.write('\n');
  }
  
  OSCCAL = original_osccal;
}

void loop() {
}
```

MCUdude, 2024-10-25:
Great to hear you figured it out!

closed 2024-10-25 by MCUdude