MCUdude/MiniCore #323: avrdude supplied by arduino ide does not support -curclock (urboot), making latest versions of minicore unusable

hMihaiDavid, 2024-11-22:
Hello.
I have Arduino IDE 2.3.3 (cli version 1.0.4), recently downloaded from official site.

I have a at168pa mini-like board and I recently flashed it with urboot, using avrdude 7.x from debian 12 I can -curclock -t and read some memory to confirm that bootloader is installed and apparently working correctly.

In Arduino IDE trying to upload the blink example, we see 
```
avrdude: Can't find programmer id "urclock"

Valid programmers are:
  2232HIO          = FT2232H based generic programmer
... ... ...
```

Arduino IDE seems to use avrdude version 6.x, which, after checking, does not have support for urboot.
```
~/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino17/bin/avrdude
```

Appreciate any help on how can I get this to work. Is there a way to tell arduino ide to use a different avrdude binary?

NOTE: Symlinking the path shown above to avrdude to the system install one seems to give same error despite the system avrdude supporting urboot.... So I do not know what's going on, probably I'm doing something wrong.

MCUdude, 2024-11-22:
You need to use MiniCore v3 or newer. I just released MiniCore v3.0.3, which bundles a bunch of Urboot v8.0 bootloaders and Avrdude v8.0. Install the latest MiniCore through the Boards manager. That will fix the Avrdude issue

hMihaiDavid, 2024-11-22:
@MCUdude After installing using the board manager it works perfectly. Seems like the manual instalation (unziping master.zip into ~/Arduino/hardware) is broken. I changed title.

Thank you!

closed 2024-11-24 by hMihaiDavid