MCUdude/MiniCore #324: How to select the bootloader variant

elogh, 2024-12-15:
Hi Hans,

is there a way to select the bootloader to be burned by MiniCore?
Or, if not, is it ok to replace the one burned for distinct settings by the (properly renamed) bootloader of choice?
(I ask because I want to burn a dual-boot variant from 
...\packages\MiniCore\hardware\avr\3.0.1\bootloaders\urboot\atmega328p\watchdog_1_s\autobaud\uart0_rxd0_txd1\template_dual
(after patching).
Thanks,
Erwin

MCUdude, 2024-12-15:
It's not possible using Arduino IDE, but I think it's possible to specify the bootloader file when using PlatformIO.
But it is trivial to just run Avrdude manually.

If you turn on verbose upload in the IDE settings, you can copy the Avrdude command and modify this to flash the bootloader file you want to use instead

elogh, 2024-12-16:
Ok, thanks!