MCUdude/MiniCore #327: Debug support for Arduino IDE 2 is working!

felias-fogg, 2024-12-30:
If you want to debug your MiniCore sketches on real hardware, you can do that now using [dw-link](https://github.com/felias-fogg/dw-link/). A [quick-start guide](https://github.com/felias-fogg/dw-link/blob/V4/docs/quickstart-Arduino-IDE2.md) shows you the necessary steps.

MCUdude, 2024-12-30:
Very, very cool! Sorry for noe being more up-front. I'll definitely give it a try in early January, and ideally add a/your quick-start guide to the MiniCore readme

felias-fogg, 2024-12-30:
Hi Hans, 

It would be cool if you could include the modifications to MiniCore. I can provide a PR for the current version. I could also produce a new debug-enabled version for each of your new  MiniCore versions.

The modification now includes new tools (avr-gdb and a gdbserver), which are not yet in their final version, though. avr-gdb is copied from various sources but not statically linked. The gdbserver is a Python program that is "compiled" using pyinstaller. Both will probably work most of the time. But they are large and compatibility is sort of unsure.

I plan to use cross-compiling avr-gdb statically linked and write a minimal gdbserver in C++, which also could be cross-compiled statically linked.  

Cheers,
Bernhard

closed 2025-05-01 by felias-fogg