MCUdude/MiniCore #333: ATmega-328pb does not boot load with arduino isp

shanmugamIoT, 2025-02-10:
I have a custom board using with atmega-328bp micro controller and i try to bootload using with arduino-isp along with your package but could not bootloads at a lot of attempts .i am so frustating about that so expers please give me a suggestions

MCUdude, 2025-02-10:
I suggest you paste the entire error message Avrdude gives you.

shanmugamIoT, 2025-02-10:
home/siva/.arduino15/packages/MiniCore/tools/avrdude/8.0-arduino.1/bin/avrdude -C/home/siva/.arduino15/packages/MiniCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf -v -patmega328pb -cstk500v1 -P/dev/ttyACM0 -b19200 -e -Ulock:w:0xff:m -Uefuse:w:0b11110110:m -Uhfuse:w:0xd7:m -Ulfuse:w:0b11111111:m 
Avrdude version 8.0-arduino.1
Copyright see https://github.com/avrdudes/avrdude/blob/main/AUTHORS

System wide configuration file is /home/siva/.arduino15/packages/MiniCore/tools/avrdude/8.0-arduino.1/etc/avrdude.conf
User configuration file /home/siva/.avrduderc does not exist

Using port            : /dev/ttyACM0
Using programmer      : stk500v1
Setting baud rate     : 19200
AVR part              : ATmega328PB
Programming modes     : SPM, ISP, HVPP, debugWIRE
Programmer type       : STK500
Description           : Atmel STK500 v1
HW Version            : 2
FW Version            : 1.18
Topcard               : Unknown
Vtarget               : 0.0 V
Varef                 : 0.0 V
Oscillator            : Off
SCK period            : 0.0 us
XTAL frequency        : 7.372800 MHz

AVR device initialized and ready to accept instructions
Device signature = 00 00 00 (retrying)
Device signature = 00 00 00 (retrying)
Device signature = 00 00 00
Error: invalid device signature
Error: expected signature for ATmega328PB is 1E 95 16
  - double check connections and try again, or use -F to carry on regardless

Avrdude done.  Thank you.
Error while burning bootloader.

MCUdude, 2025-02-10:
```
AVR device initialized and ready to accept instructions
Device signature = 00 00 00 (retrying)
Device signature = 00 00 00 (retrying)
Device signature = 00 00 00
```

The Arduino as ISP programmer can't communicate with the ATmega328PB target. Do you have a schematic you can share? It may be because no external clock/crystal is connected to the 328PB, or it's not oscillating. I always recommend getting a proper ISP programmer such as the USBasp, because there are so many hardware-related issues when users use the Arduino as ISP.

ANSB, 2025-04-22:
Hi,
I have the same problem and I discovered something surprising. A friend has been using MiniCore for quite some time, while I only recently installed it. When searching for information or videos on using MiniCore, you see that when you choose a chip (for example, MiniCore->ATMega-328) in the Arduino IDE's programmer selection menu, the available programmers all have "(MiniCore)" written after their name.
At least, that's what you see in the videos, and that's what my friend has on his IDE. But not me. In the list, only the STK500 is noted as "(MiniCore)," but not the others. So I think that only the STK500 is considered to have a Bootloader available for it, while the other programmers are considered by the IDE as not having the necessary file. Perhaps this explains why installing the bootloader with "Arduino as ISP" fails all the time?
I'm using a Mac with Sequoia 15.4.1, the Arduino IDE is the latest (1.8.19), and the MiniCor package is 3.0.4 installed via the IDE.

MCUdude, 2025-04-22:
> When searching for information or videos on using MiniCore, you see that when you choose a chip (for example, MiniCore->ATMega-328) in the Arduino IDE's programmer selection menu, the available programmers all have "(MiniCore)" written after their name.

The `(MiniCore)` suffix was there to distinguish the different programmers back when the IDE placed app programmers for all Arduino cores in a single list. This is now handled differently. I've removed `(MiniCore)` because it's not necessary anymore. This will take effect in the next MiniCore release.

If you're having problems using the Arduino as ISP programmer, please turn on verbose compilation and upload in the IDE settings and post the entire output here. Only then I'm able to help out.

In general, the Arduino as ISP itself is usually the root cause for the problems. I would definitely have fewer reported issues if I simply removed this option in the Programmers menu and suggested getting a "proper" programmer like the USBasp or the USBtinyISP.