MCUdude/MiniCore #334: Atmega168 error "avrdude: Can't find programmer id "urclock""

Parksy1234-cloud, 2025-02-12:
What started as bad bootloader problem on an arduino nano has got me into deeper doo-doo after trying to resolve it by using minicore, cant seem to find any help anywhere with this error.
I thought it was just lack of support in arduino ide.
Minicore did help in me in writing a bootloader for it, and told me the id sig was actually a 168 not a 328, but on compiling and then trying to upload i have become unstuck with this "urclock" error while using usb

[error.txt](https://github.com/user-attachments/files/18760763/error.txt)

 to upload it.

MCUdude, 2025-02-12:
You need to install the latest MiniCore version though Boards manager. Arduino IDE is trying to use Avrdude v6.3, while MiniCore and ``urclock`` requires v8.0

Parksy1234-cloud, 2025-02-14:
Thank you MCUdude, that fixed it, then with only 30 lines of code it ran out of memory, had to make do with 150 FastLED's instead of 200, Cheers.

MCUdude, 2025-02-14:
Yes, the ATmega168 has minimal RAM, so you'll hit the wall quickly when working with addressable LEDs. Great to hear you figured out the Avrdude issue!

closed 2025-02-14 by MCUdude