MCUdude/MiniCore #337: 328PB Circuit Validation

Riz-waan, 2025-03-15:
Hi Hans,

I am working on a custom Arduino Pro Mini based on the ATMEGA328pb for a school project. I'm just not understanding section 11.3 of the datasheet: [11.3 Low-Power Crystal Oscillator](https://onlinedocs.microchip.com/oxy/GUID-0EC909F9-8FB7-46B2-BF4B-05290662B5C3-en-US-12.1.1/GUID-B8CD4B8B-3A7F-4649-ABA1-751A2849D783.html). 

Me and my professor are confused by all of the equations, specifically the one that says Ce + Ci + Cs <= 22pF but since Ci is 18pF of XTAL1, does that mean we cannot use an external capacitor at all? Also the Ce + Ci = 2CL - Cs equation.

Also, would you suggest just using the murata ceramic resonator with the integrated 15pF since I have seen a lot of people use it on here? I did want to use a crystal oscillator for better timing but am so confused by the datasheet. In addition, don't you have to set some fuses or something if you are using external ceramic resonator vs crystal as per the datasheet of ATMEGA328pb?

Finally, I know I do not have the diode for the reset pin, not sure what value to use. 

[image]

I would appreciate any guidance you can provide.

MCUdude, 2025-03-15:
Finding a suitable crystal can be difficult. I've had lots of trouble with this before. My best advise is to get a clock generator instead. It outputs a square wave that you connect directly to the XTAL1 pin on the 328PB. I've used the YXC OT32258MJBA4SL (available on LCSC) for a product that had to run at 8 MHz. A suitable 16 MHz option is probably available as well.

MCUdude, 2025-03-15:
Also, use a diode on the reset line. It will make uploading more reliable. Definitely worth it IMO. Any fast signal diode or Schottky would do.

Riz-waan, 2025-03-15:
Awesome, thanks for your suggestion. So just to confirm, I would still power the YXC with 3.3V and its output would still be read by the atmega328pb at 5v?

Any chance you could email me at gmc896@mocs.utc.edu.

closed 2025-03-15 by Riz-waan

MCUdude, 2025-03-16:
> So just to confirm, I would still power the YXC with 3.3V and its output would still be read by the atmega328pb at 5v?

Yes, correct. The system I used it on was running at 3.3V, so that's why I chose that particular clock generator. But since your system is running at 5V, you may be able to find one that can run on 5V, so you don't need a 3.3V regulator.

Another alternative is to use a very spesific resonator with built-in capacitors. I've used the Murata CSTNE8M00G520000R0, but I think I'd stick with a clock driver due to all the crystal related issues that's being reported here.