MCUdude/MiniCore #338: ATMEGA 48-20U and ATMEGA 48-PA-PU flash and upload successful but don't run sketches

friek1, 2025-04-07:
I've been trying out several ATMEGA chips with MiniCore and I've run into issues with the ATMEGA48 variants. They successfully flash with the bootloader and let me upload sketches, but they don't execute any of those sketches.

What I've tried:
* Wrote a minimal AVR-GCC blink example and flashed the chip with avrdude which worked fine. (Just the 48-20U)
* Flashed the Minicore-Bootloader => No Errors
* Uploaded a serial hello world sketch => No output
* Uploaded a Blink Sketch => No output

The chips I've tried:

* ATMEGA 8A-PU => _Worked_
* ATMEGA 48-20U => **No sketch execution**
* ATMEGA 48-PA-PU => **No sketch execution**
* ATMEGA 88A-PU => _Worked_
* ATMEGA 168-20 => _Worked_
* ATMEGA 328-P => _Worked_

I'm using 
* Arduino **IDE 2.3.4**
* comes with avrdude version **6.3** in the **Arduino9** directory
* comes with avrdude version **6.3-20190619** in the **Arduino17** directory
* avrdude version **7.1** on my system, **LinuxMint**
* Arduino **Nano as ISP**

If I've overlooked any documented incompatible chip variants, I would like to apologize in advance. Otherwise I'd appreciate any advice on how to further debug this issue.

MCUdude, 2025-04-07:
Are you using the latest MiniCore version (v3.0.3), installed though the boards manager? Any ATmega48 should work, so it could be a bug somewhere. Can you provide your example sketch so that I can try the exact same thing as you.

friek1, 2025-04-07:
Hi, yes, I'm using v3.0.3 from the board manager.

My example somehow ended up always being this:

int comp;
void setup() {
  Serial.begin(115200);
}

void loop() {
  Serial.println("Hello world!!!!");
  Serial.println(millis());

  comp = millis()%1000;
  delay(1000-comp);
}

friek1, 2025-04-07:
I just re-tried burning the bootloader and uploading the sketch with Arduino IDE 1.8.19

* Burning the bootloader works
* Uploading a sketch gives the following output:

> Warning: platform.txt from core 'MiniCore' contains deprecated compiler.path={runtime.tools.avr-gcc.path}/bin/, automatically converted to compiler.path=/usr/bin/. Consider upgrading this core.
> Sketch uses 1692 bytes (44%) of program storage space. Maximum is 3776 bytes.
> Global variables use 108 bytes (21%) of dynamic memory, leaving 404 bytes for local variables. Maximum is 512 bytes.
> Warning: no eeprom data found in Intel Hex file /tmp/arduino_build_447593/uarthelloworld.ino.eep
> Warning: flash verification mismatch
>   device 0xff != input 0x3e at addr 0x0002 (error)
> Error: flash verification mismatch
> Error: flash verification mismatch

Arduino IDE 2.3.4 still gives no error and tells me that the upload is complete.

closed 2025-04-07 by MCUdude

MCUdude, 2025-04-07:
I did find a bug! For some reason, I've forgotten to set the `selfprogen` fuse. f38614e resolves this, and a new boards manager version, v3.0.4 will be released soon.

MCUdude, 2025-04-07:
v3.0.4 is not available though boards manager!

friek1, 2025-04-08:
Thanks much! :)