MCUdude/MiniCore #347: Compile error in new MiniCore 3.1.0

Nicks7777, 2025-05-01:
After installing the new version of MiniCore 3.1.0, sketches stopped compiling.

Before:

Sketch uses 31966 bytes (98%) of program storage space. Maximum is 32384 bytes.
Global variables use 556 bytes (27%) of dynamic memory, leaving 1492 bytes for local variables. Maximum is 2048 bytes.

After:

C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:2: error: stray '\5' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
  ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:3: error: stray '\26' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
   ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:4: error: stray '\7' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
    ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:6: error: stray '\2' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
      ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:26: error: stray '\2' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                          ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:42: error: stray '\2' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                          ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:46: error: stray '\243' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                              ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:96: error: stray '\243' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:100: error: stray '\230' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                    ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:120: error: stray '\1' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                        ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:124: error: stray '\230' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                            ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:131: error: stray '\25' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                                   ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:153: error: stray '\1' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                                                         ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:154: error: stray '\2' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                                                          ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:156: error: stray '\303' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                                                            ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:157: error: stray '\34' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                                                             ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:158: error: stray '\317' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                                                              ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:160: error: stray '\255' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                                                                ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:161: error: stray '\241' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                                                                 ^
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\._Wire.cpp:1:9: error: 'Mac' does not name a type
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
         ^~~
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\utility\._twi.c:1:2: error: stray '\5' in program
[... 41 lines cut ...]
[... cut ...]
C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0\libraries\Wire\src\utility\._twi.c:1:161: error: stray '\241' in program
     Mac OS X                2   q      �                                      ATTR       �   �                     �     com.apple.provenance  ��j��uZ
                                                                                                                                                                 ^

exit status 1

Compilation error: exit status 1

MCUdude, 2025-05-01:
Weird. I've never seen this before. Which operating system are you using?

Nicks7777, 2025-05-01:
1. Windows 7 Ultimate
2. The same thing happens

Can't upload screenshots and sketch

MCUdude, 2025-05-01:
What does it looks like if you enter the `C:\Users\Admin\AppData\Local\Arduino15\packages\MiniCore\hardware\avr\3.1.0` folder? Are there a bunch of files starting with `._`? Whar happends if you delete the files that starts with `._` and the compiler complains about?

I'm trying to get to the bottom of this. I can't see these files on my computer...

Nicks7777, 2025-05-01:
this is what is installed since 3.1.0, and so on almost all kernel folders.

02.05.2025  08:33    <DIR>          .
02.05.2025  08:33    <DIR>          ..
02.05.2025  08:33               163 ._utility
02.05.2025  08:33               163 ._Wire.cpp
02.05.2025  08:33               163 ._Wire.h
02.05.2025  08:33               163 ._Wire_timeout.h
02.05.2025  08:33    <DIR>          utility
02.05.2025  08:33            22 517 Wire.cpp
02.05.2025  08:33             6 318 Wire.h
02.05.2025  08:33               105 Wire_timeout.h

Nicks7777, 2025-05-01:
after removing all files from the kernel, the "_*.*" sketch compiles normally

MCUdude, 2025-05-01:
Good. I've just updated the 3.1.0 compressed file that Arduino IDE downloads, and I've updated the boards manager file. Please give it an hour or so for the server to catch up. I've seen boards manager related delays before. in the meantime, delete your Minicore 3.1.0 installation, including the dowloaded archive the IDE also saves

Nicks7777, 2025-05-01:
I think it would be better to make version 3.1.1

MCUdude, 2025-05-03:
V3.1.1 is now available though boards manager

closed 2025-05-03 by MCUdude

Nicks7777, 2025-05-03:
OK