MCUdude/MiniCore #354: Restoring UNO to its original state after debugging

felias-fogg, 2025-06-01:
I try to write an easy-to-read description on how to restore an UNO R3 after a debugging session. It essentially involves closing the open solder bridge (or re-soldering a removed cap) and then flashing the bootloader. 

If the user is happy with MiniCore and urboot, everything is easy. I am wondering whether it would be possible to make things even smoother. Avrdude can now send a pulse over the DTR line, right? Couldn't that be used to reset the board when one does not have a reset capacitor? This would require some modifications to avrdude.conf, right?  Do you think this is feasible, or could that interfere with other things in the Arduino IDE, such as the monitor? If one could implement it, it would be a huge incentive to move to MiniCore, as the RESET cap would then be superfluous. 

If the user wants to switch back to Optiboot for whatever reason, then things become a bit messy. The problem is that the standard "Arduino AVR Boards" package does not know all the debuggers (Power debugger, PICkit4, and SNAP are missing). And since they are using avrdude 6.3, one gets an (inconsequential) error message when the lock bits are written with Atmel-ICE. So the only general advice I can give is to use any ordinary ISP programmer or Arduino as ISP to burn the bootloader.  And I guess, this is what I will do. 

Two alternatives to the previous solution are:
- Setting up a fork of the Arduino AVR Boards as "Arduino AVR Boards (Debug enabled)", which has all the relevant debuggers in the programmer list and uses avrdude 8.0
- Permitting to flash optiboot again in MiniCore

I guess both of these solutions are a bit too much, and you probably dislike the second solution.

What is your take of all of this?

MCUdude, 2025-06-02:
> I am wondering whether it would be possible to make things even smoother. Avrdude can now send a pulse over the DTR line, right?

Well, yes. But this depends on which programmer is in use. It sends a pulse if you're using `-c urclock` or `-c arduino` at least.

> Couldn't that be used to reset the board when one does not have a reset capacitor?

More recent versions of Avrdude toggles the DTR line in a way that allows for both a 100nF capacitor in series, and a direct connection between DTR and the reset line, given that the two operates at the same voltage levels. Here are the DTR logic for Urboot. Note that `(&pgm->fd, 0)` means that the line goes high (to Vcc) and `(&pgm->fd, 0)` means that the line goes low (to GND)
https://github.com/avrdudes/avrdude/blob/231bb28db58fe8a016b58526514510ed6fb03135/src/urclock.c#L2269-L2276

> Arduino as ISP

If I understand you correctly, you want to suggest using an Arduino as ISP, but without the 10uF capacitor if possible? Arduino as ISP uses `-c stk500v1` when communicating with Avrdude. When looking at the [stk500v1 Avrdude implementation](https://github.com/avrdudes/avrdude/blob/main/src/stk500.c), it looks like the DTR/RTS line isn't touched at all by Avrdude, unless you're using `-c arduino`, which is what Optiboot bootloaders are using.

But even though Avrdude isn't touching the DTR/RTS line, the driver itself may pull this line high or low depending on how it is configured. Maybe we could add the same extended parameter option as the SerialUPDI programmer has, an option to force the DTR/RTS line high or low during the entire session? This would remove the need for a 10uF reset capacitor on the Arduino as ISP. But we would have to hurry up then. Avrdude 8.1 will be released in approximately. 14 days!

MCUdude, 2025-06-02:
Regarding the Arduino as ISP. If you use Avrdude 8.0, you may be able to use `-c arduino -b 19200` to communicate with the Arduino as ISP programmer. It will reset the programmer using the DTR line, but it will give the programmer 10 sync attempts before giving up. As of right now, `-c stk500v1 -b 19200` or `-c arduino_as_isp` gives you no additional sync attempts, so a DTR pulse will disrupt the communication.

* Please try `-c arduino -b 19200` to see if this works (without the 10uF capacitor)
* I'll see if we can add optional auto reset functionality, and more importantly, more retry attempts for the stk500v1 and Arduino as ISP. Ideally, before the 8.1 release

MCUdude, 2025-06-02:
> Permitting to flash optiboot again in MiniCore

It is difficult to know exactly what kind of bootloader is present. Why not dump the entire flash, save it as a file, and write it back after the debug session has ended? This is trivial to do with Avrdude.