MCUdude/MiniCore #355: MiniCore Upload Works in Arduino IDE but Fails in VSCode

parkkh12, 2025-06-20:
Hi! I'm trying to use MiniCore with VSCode. Uploading works fine in the Arduino IDE (using UART), but it doesn't work in VSCode. Is there something I might be missing?

[image]

ritesh006, 2025-08-25:
is this still relevant then please let me know how reproduce this bug

ritesh006, 2025-09-04:
[env:uno_minicore]
platform = atmelavr
board = uno
framework = arduino
board_build.core = MiniCore
board_build.mcu = atmega328p
board_build.f_cpu = 16000000L
upload_protocol = arduino
upload_port = COM4
upload_speed = 115200
monitor_speed = 115200

This settings works for me may be for you also it will work